TOPLLab/WARDuino #1: Source-level debugging?

vshymanskyy, 2020-06-14:
Debugging wasm opcode execution doesn't seem very useful. Is source-level debugging available? If yes, are there any docs on this?

beardhatcode, 2020-06-15:
Source level debugging is not yet availible, we are vorking on this right now.

It can already be done (painfully) by hand:

1. get the source maps from emscripten/...
1. manually find the correct byte offset. 
1. Send `10` over the serial port to obtain the start position of the WA code in the VM,
1.  add the offset and 
1. set a breakpoint there.

vshymanskyy, 2020-06-18:
I decided to publish my WIP debugger, that I develoepd some time ago while working on Wasm3. Check it out!
https://github.com/wasm3/wasm-debug
Maybe we could collaborate in this direction

beardhatcode, 2020-06-18:
I have just sent you a mail to ask if we could collaborate. (I got your mail from your commits)

vshymanskyy, 2021-10-19:
Not sure if the referenced commit resolves the issue.