TOPLLab/WARDuino #4: WASI support

beardhatcode, 2021-02-22:
We should look into supporting WASI.

carllocos, 2021-02-22:
Note to ourselves: 
in case we want to test the WASI using emscripten, make sure to use at least v1.39.5. Since then, emscripten assumes WASI support by the runtime.

Also since [emscripten v1.38.47](https://emscripten.org/docs/introducing_emscripten/release_notes.html) a standalone compilation to wasm no longer works with the `-s ONLY_MY_CODE=1` option. We need to use `-s STANDALONE_WASM` instead.

beardhatcode, 2021-03-27:
@carllocos , I have been digging a bit into emcc and the following works for me with EMCC 2.0

```
	emcc FILE \
		-s STANDALONE_WASM \
		--ignore-dynamic-linking \
		-s WASM=1 \
		-s NO_FILESYSTEM=1 \
		-s SIDE_MODULE=1 \
		-O3 \
		-o $@
```

carllocos, 2021-03-28:
I sincerely do not remember how I managed to compile WASM that relies on WASI. The script that I have, that supposedly generates WASI dependency, no longer does it... Anyways, the above works for me @beardhatcode . 

I observed that when transforming the obtained WASM to WAT from the attached file, I get imports (see below) which I'm not sure whether WARDuino provides. Just mentioning it for safety.

to WAT command:
`wasm2wat game.wasm -o game.wat`

The imports:
   `(import "env" "stackSave" (func (;0;) (type 1)))`
   `(import "env" "memory" (memory (;0;) 0))`
   `(import "env" "__indirect_function_table" (table (;0;) 0 funcref))`

[game.txt](https://github.com/TOPLLab/WARDuino/files/6217579/game.txt)