TOPLLab/WARDuino #9: How to build WARDuino for other mcu's

morsedev, 2021-03-16:
Hi! I would like to know if there is a list of supported mcu's other than esp32/esp8266. If not, it is possible to get some instructions on how to build WARDuino for other Arduino compatible mcu's??

Thanks.

beardhatcode, 2021-03-28:
Hi @morsedev, sorry for my late reply.

There are other supported devices, but the problem is that we need to get those devices in our hands before we can assert that they work.

In the `demo` folder there is a file called `main.esp8266.template.ino` which we use to bootstrap WARDuino. It creates the module and starts the computation and debugging facilities. You can try to clone this repository under `$(HOME)/Arduino/libraries` and open a copy of the ino fine in the Arduino IDE. And push it to you device to see if it works.

https://github.com/TOPLLab/WARDuino/blob/3f349ed7e9506b1f00113ba1375520294d91608e/demo/main.esp8266.template.ino#L46-L53

There are a lot of things taht can go wrong with a new device starting with having the wrong pin number. This [hello_world.ino](https://github.com/TOPLLab/WARDuino/files/6217670/hello_world.txt)  file uses pin number  16 for the led pin.

https://github.com/TOPLLab/WARDuino/blob/3f349ed7e9506b1f00113ba1375520294d91608e/demo/run_wa/wa_sources/hello_world.wast#L24-L30

If that is an other pin you'll need to adjust that in that wast file and run `make hello_world.ino` in demo/run_wa. Once you get it set up with the arduino IDE,you can update [scripts/upload](https://github.com/TOPLLab/WARDuino/blob/3f349ed7e9506b1f00113ba1375520294d91608e/scripts/upload) to also support your device.

Do let us know how it goes.

tolauwae, 2022-04-06:
Closing this stale issue.

Since the project overhaul #33, the top level [README](https://github.com/TOPLLab/WARDuino#readme) contains more detailed instructions on how to run the VM on embedded devices.

closed 2022-04-06 by tolauwae