TOPLLab/WARDuino #18: step does not advance when bp reached

carllocos, 2021-04-10:
Since the solution on [issue 5](https://github.com/TOPLLab/WARDuino/issues/5), the "step" interrupt no longer advances after reaching a breakpoint . The cause is the following if 

[https://github.com/TOPLLab/WARDuino/blob/3f349ed7e9506b1f00113ba1375520294d91608e/instructions.cpp#L1498](https://github.com/TOPLLab/WARDuino/blob/3f349ed7e9506b1f00113ba1375520294d91608e/instructions.cpp#L1498)

when a breakpoint is reached and you step the branch condition evaluates to true.

beardhatcode, 2021-04-10:
Would this be solved by also putting these lines in the step case?\

https://github.com/TOPLLab/WARDuino/blob/3f349ed7e9506b1f00113ba1375520294d91608e/interrupt_operations.cpp#L257-L260

carllocos, 2021-04-10:
Yes certainly! I'll make a PR later for that fix.

tolauwae, 2022-03-31:
I believe this has been merged. Closing issue.

closed 2022-03-31 by tolauwae