TOPLLab/WARDuino #48: Interrupt payload needs to end with a space and a new line

carllocos, 2022-04-13:
I'm writing a python client to WARDuino and noticed that when sending an interrupt through the socket, the interrupt payload has to end with a space and a new line in order to be processed by WARDuino: `'60 \n'` works fine but `'60\n'` or `'60 '` does not work.

However, when sending interrupts through Telnet the space is not needed i.e., it suffices to type `'60'` and `press enter`.

beardhatcode, 2022-04-13:
Have you tried forcing pyton to flush its buffer?

tolauwae, 2022-04-20:
Still haven't tracked down the source of this bug. But for the record, the plugin also needs to send a space.

carllocos, 2023-04-14:
I found and fixed the problem See https://github.com/TOPLLab/WARDuino/issues/122