TOPLLab/WARDuino #73: AssemblyScript example optimized version fails

tolauwae, 2022-06-01:
Trying to run the AssemblyScript example without the mqtt fails. (= programming demo example)

When the busy loop is empty, binaryen optimizes away callback functions. It eliminates the `funcref` table and passes the function's own index to the `subscribe_interrupt` primitive.

```typescript
    while (true) {    
        wd.sleep(5); // Sleep for 5 seconds    
    }
```

The resulting index is obviously out of range since there is no table. This causes an unhandled exception whenever the VM wants to resolve an event.

It seems users of Asyncify have run into similar problems: https://github.com/WebAssembly/binaryen/issues/4484

carllocos, 2023-07-13:
I also experience this issue, however, in my case, the table is not compiled away but the table index is clearly out-of-range

tolauwae, 2023-07-13:
The table is never compiled away again. I fixed that issue a while back by setting the correct compiler options.

But callbacks still don't work, because the table index is passed through wasm memory now instead of directly (like before). See PR #197 for a temporary fix.