TOPLLab/WARDuino #75: Update smartlamp.ts example

louisvangeldrop, 2022-06-05:
I am unable to connect to Wifi when using the smartlamp.ts example.
However if I change the following function:

function until_connected(connect: () => void,
                         connected: () => boolean): void {
    while (!connected()) {
        wd.delay(1000);
        connect();
    }
}

into 

function until_connected(connect: () => void,
                         connected: () => boolean): void {
    while (!connected()) {
        connect();
        wd.delay(5000);
           }
} 

the program connects to Wifi.

tolauwae, 2022-06-07:
Hey @louisvangeldrop

Thanks for the issue! We found this problem a few days ago as well.
Indeed, we retry connecting too quickly, so the board can never connect.

We fixed this on a branch that still needs merging: https://github.com/TOPLLab/WARDuino/blob/fb76fffb811726ba481b0011a98b1845f4c90191/examples/assemblyscript/main/smartlamp.ts#L9-L16

This code will wait until there is a wifi connection, but only call `connect` again when the previous call has exhausted its maximum number of attempt. `wd.wifi_status` returns `1` while trying to connect and `6` if it has failed or disconnected.

_(This is also the proper way of using the underlying Arduino implementation, where `Wifi.Begin` shouldn't be called before status returns `WL_IDLE_STATUS`. ([arduino docs](https://www.arduino.cc/reference/en/libraries/wifi/)))_

louisvangeldrop, 2022-06-07:
Hallo @tolauwae,

Thanks for the update. I will copy your code into my project and use it. 

Now I have got the wifi-connection working, it is time to find out why the Mqtt will not connect. 
It returns State = -2.

closed 2025-01-21 by tolauwae