TOPLLab/WARDuino #96: resole_event does not modify the value_type of pushed stack values

carllocos, 2022-10-06:
In `resolve_event` the values pushed to the stack do not get assigned any value_type. This causes the VSCode plugin to fail.  https://github.com/TOPLLab/WARDuino/blob/631abafef6e9df4ca7963370c5b53fdabc25886d/src/WARDuino/CallbackHandler.cpp#L105

tolauwae, 2022-10-06:
Probable fix in `Callback::resolve_event`:

```c
    // Push arguments (5 args)
    uint8_t uint32 = 0;
    module->stack[++module->sp].value.uint32 = start - topic.length();
    module->stack[module->sp].value.uint32 = uint32;

    module->stack[++module->sp].value.uint32 = topic.length();
    module->stack[module->sp].value.uint32 = uint32;

    module->stack[++module->sp].value.uint32 = start;
    module->stack[module->sp].value.uint32 = uint32;

    module->stack[++module->sp].value.uint32 = payload.length();
    module->stack[module->sp].value.uint32 = uint32;

    module->stack[++module->sp].value.uint32 = payload.length();
    module->stack[module->sp].value.uint32 = uint32;
```

carllocos, 2023-04-14:
Indeed. So not assigning the type to the stack values was the problem. This bug has been fixed in the meantime.

closed 2023-04-14 by carllocos