TOPLLab/WARDuino #97: Generating events rapidly crashes the VM

carllocos, 2022-10-06:
When pressing buttons, on which callback handlers are registered, rapidly after each other causes the VM on the MCU to crash. This behaviour does not occur that often and the reason may be that the event queue limit is reached.

tolauwae, 2022-10-10:
The event queue is only limited by the memory of the device. The events for button presses are so small, you would have to generate an unreasonable amount to run out of memory on the device. (provided you program isn't already running against the limit)

I can't replicate this problem at all.
When pushing the button repeatedly on my device, in very quick succession, the mcu never crashes.

@carllocos Could you provide more information on this? Or check if this is indeed still a thing?

carllocos, 2023-04-14:
This no longer happens. I believe some fix that you made some time ago solved the problem. I would like to refer to the fix commit for proper documentation but unfortunately, I do not find it.

Closing the issue

closed 2023-04-14 by carllocos