TOPLLab/WARDuino #117: Loading Factorial module gives a Parsing type error

carllocos, 2022-12-02:
### Context
So I'm currently implementing the update module feature (**branch feat/update-module** 5e29643ea48dd08a7462ad052e4478090c8f36f9). To this end, I refactored a part of the behaviour of ``WARDuino::unload_module`` and ``WARDuino::load_module`` into methods ``WARDuino::load_module_state`` and ``WARDuino:: free_module_state``

- ``free_module_state`` will free the callstack, stack, globals, etc.
- ``load_module_state`` is now the one that parses wasm bytes and allocates the callstack, stackvalues, etc.

Implementation wise I just moved a big chunk of ``WARDuino::unload_module`` into   ``free_module_state`` and ``WARDuino::load_module`` into ``WARDuino::load_module_state``, which can be seen in commit d84cf0554be0fed90f898109732198ed220c8918.

### Issue

Loading the state of a simple [factorial wasm module](https://github.com/TOPLLab/WARDuino/blob/feat/module-update/tests/vm_unit_tests/example_code/fac/fac.wast) as shown in: https://github.com/TOPLLab/WARDuino/blob/5e29643ea48dd08a7462ad052e4478090c8f36f9/tests/vm_unit_tests/statemodule_test.cpp#L43

Triggers an assert failure ``WARDuino/WARDuino.cpp:364): 0 -th type def was not a function type`` on line https://github.com/TOPLLab/WARDuino/blob/5e29643ea48dd08a7462ad052e4478090c8f36f9/src/WARDuino/WARDuino.cpp#L362

I probably did a mistake while refactoring the code to above methods but when running the code with the [blink module](https://github.com/TOPLLab/WARDuino/blob/feat/module-update/tests/vm_unit_tests/example_code/blink/blink.wast) the issue does not arise.

I also executed the factorial wasm with the emulator of the develop branch through command ``./wdcli --file fac.wasm`` but there I get a segmentation fault, whereas blink works without any issue.

Any idea @tolauwae ?

tolauwae, 2022-12-02:
**1.** I'm a bit confused by the split of the functions and the naming of the new functions.

Firstly, I don't see why the split is functionally necessary. Because I don't think you can ever call `load_module_state` on its own. For instance, I think that the `module::update_module` function won't work because it doesn't add the module to the available modules in WARDuino (`warduino->modules`).

https://github.com/TOPLLab/WARDuino/blob/5e29643ea48dd08a7462ad052e4478090c8f36f9/src/WARDuino/WARDuino.cpp#L964

**Important to note:** support for multiple modules hasn't been officially added, this code is simply "future proof"

That being said, I do think the split might be good to make the code more readable. However, the naming isn't self-documenting enough and at first glance the new function should actual be named `load_module` and the old method should be renamed to something like `import_module` since it adds the module to the lists of modules available to WARDuino.

**2.** With regard to the error you get, if I had to bet this is caused by a standardised Wasm spec proposal that WARDuino doesn't support yet. Since, the spec test still succeeds in the CI on you branch it seems the load module works fine for that suite. So this is probably an issue with you compiler using a Wasm proposal we don't yet support. Does the suite succeed locally? How is the fac program generated?

**3.** Lastly, it sounds as though the seg fault should be a separate issue.

tolauwae, 2022-12-03:
Ok nvm this comment:
> Firstly, I don't see why the split is functionally necessary. Because I don't think you can ever call `load_module_state` on its own. For instance, I think that the `module::update_module` function [...]

I just realised you're literally replacing the loaded module. So this should probably work.

tolauwae, 2022-12-05:
@carllocos I had a closer look at your problem and did some debugging of my own. There is an issue with your implementation of factorial.

At each iteration of the loop the result is kept on the stack and never gets removed. It appears to me as though the stack fault is caused by a full stack. The vm should probably detect this and give a clear error message, rather than try to access memory out of bounds.

This should be fixed on a different branch and merged as soon as possible to develop. More than happy to review the PR :)

https://github.com/TOPLLab/WARDuino/blob/5e29643ea48dd08a7462ad052e4478090c8f36f9/tests/vm_unit_tests/example_code/fac/fac.wast#L29-L32

tolauwae, 2022-12-05:
I added a fixed version of the fac program to the examples folder: ba530d729c03d8eca006a786af9961e564f65346

Also made a separate issue for the seg fault #118. Could you follow up on this @carllocos?

So this issue can be closed now.

closed 2022-12-05 by tolauwae