TOPLLab/WARDuino #119: Double free when updating module

tolauwae, 2022-12-09:
I use the following command (`0x22`) to update the module of a running WARDuino instance:

```bash
xxd -p hello_world.wasm | tr -d "\n" | sed "s/^/220000007a/" | sed "s/$/\n/" | netcat localhost 8300 -q -1
```
*(The size of the module is 122 = `0x0000007a`)*

But I get a double free when `arecalloc` is called in `WARDuino::load_module_state WARDuino.cpp:484`
https://github.com/TOPLLab/WARDuino/blob/37eb658f48da223f4a6baabf4bf3c7a393320254/src/WARDuino/WARDuino.cpp#L484-L486

I don't immediately see the cause. @carllocos is this one of the faults you are already tracking?

tolauwae, 2022-12-09:
The programming being used:
```wast
(module
 (; Arduino imports ;)

 (import "env" "print_string"       (func $print            (type $1)))
 (import "env" "chip_delay"         (func $delay            (type $2)))

 (memory $mem 1)
 (data (i32.const 0) "Hello there!")

 (; Type declarations ;)
 (type $1 (func (param i32) (param i32) (result)))
 (type $2 (func (param i32)             (result)))
 (type $3 (func (param)                 (result)))

 (; Define one function ;)
 (export "main" (func $blink_arduino))

 (; The wait function ;)
 (func $wait (type $3)
    (;  Delay time   ;)
    (i32.const 1000)
    (call $delay)
 )
 (; The blink function ;)
 (func $blink_arduino (type $3)
    (i32.const  0)
    (i32.const 12)
    (call $print)
 )
)
```

carllocos, 2022-12-10:
Indeed the double free is one of the errors that I was fixing. I just pushed a commit a6057d6bd9641a446045dbd66edf6083254ebb93 that fixes the issue.
I could not test your example here above as my netcat does not know the --q option. 

At this point, I'm confident that instantiating and freeing a module happens correctly. **However**, I still need to add tests to determine whether the update happens correctly.

closed 2022-12-10 by carllocos

tolauwae, 2023-01-05:
@carllocos I ran the example above again with `netcat` and it works perfectly. Thanks for the fix!

commit a6057d6b (2022-12-10): fix #119

Variables of module need to be nullyfied.
Otherwide, the `load_module_state` parsed the module based on stale data