TOPLLab/WARDuino #127: Fine-grained State and Update State Interrupts

carllocos, 2022-12-30:
Currently, we have several interrupts to request state from the VM (e.g., `interruptDUMP`, `interruptDUMPLocals`, `interruptWOODDUMP`, `interruptDUMPCallbackmapping`,  etc.) and others to update state (e.g, `interruptUPDATELocal`, `interruptRecvState`, etc.). It would be best to introduce only 2 interrupts such as `interruptState` and `interruptStateUpdate` to respectively request and update state and this in a coarse or fine-grained manner (e.g, request all the stack values or just one stackvalue, update the whole stack or just one stackvalue, etc.)

The idea is to play with the payload. For instance, `interruptState` could expect as payload a byte flag where each bit enables the retrieval of some possible state: `interruptState 11000` could for instance return the global variables and stack if we assume here that the first two bits respectively request for the global and stack.

The advantages of this fine-grained State and update state interrupts is mainly about optimisation: 1. tooling support becomes faster since now we can for instance update only the changed state instead of having to push the always the whole state, 2. we reduce the memory footprint on the MCU since two interrupts will encompass most of the state-related interrupts that are already present in the VM, 3. we reduce the network bandwidth that could potentially negatively affect the battery life of MCU.

The following interrupts and interrupt handlers will be definitely affected:
- interruptDUMP
- interruptDUMPLocals
- interruptDUMPFull
- interruptUPDATELocal
- interruptWOODDUMP
- interruptRecvState
- interruptDUMPCallbackmapping
- interruptRecvCallbackmapping

tolauwae, 2023-01-04:
I don't quite understand the last two advantages.

2. It isn't clear to me how this reduces the memory footprint on the MCU? Will the code be smaller? I don't believe having different interrupts introduces any duplicate code. Intuitively, I'd say the code size would be about the same.

3. How would this reduce the network bandwidth? Instead of only a single byte encoding the type of interrupt, we would have an additional payload to specify what to dump. So messages would be larger, no?

@carllocos what are your thoughts on this?

carllocos, 2023-04-14:
2. Currently the code duplication comes from having several print methods that are called by each interrupt to print the state. Despite intersections in the kind of state being printed (e.g., interruptWOODDump and interruptDump both print the callstack) we still use different print methods to print exactly the same state. The idea is simply to implement one general print method that can print any state depending on the arguments given to the method. And every interrupt could then make use of this method to print their desired state. Thus reducing code duplication and making the code more maintainable.

5. It is true that the messages would become larger due to the greater payload but that size is neglectable compared to the network bandwidth that you save when having the freedom to request only what you need. If for instance, a tool client is only interested in the global variables it could then just request the global variables by providing the right payload. As opposed to requesting a wooddump that would then print unnecessary information (e.g., memory, stack) just to get access to the globals.

Actually, in retrospect, we do not need to remove existing interrupts but behind the scene they could be reimplemented to use the same printing and state update code.

Here a first version of this idea https://github.com/TOPLLab/WARDuino/blob/1be470f049b1b3d7544636dc586c9446f4f4be42/src/Debug/debugger.cpp#L253

tolauwae, 2023-04-15:
> It is true that the messages would become larger due to the greater payload but that size is neglectable compared to the network bandwidth that you save when having the freedom to request only what you need.

Yes you're right. This is indeed a good feature. :+1:

> The idea is simply to implement one general print method that can print any state depending on the arguments given to the method.

Don't know if this has to be implemented necessarily as one giant method.

carllocos, 2023-04-16:
Of course, we can discuss on how to concretely implement it :)

MaartenS11, 2024-04-30:
Correct me if I'm wrong but this sounds like the inspect interrupt which has already been merged. If so we should close this issue.

tolauwae, 2024-05-02:
@MaartenS11 Yes, well spotted. Closing issue.

closed 2024-05-02 by tolauwae