TOPLLab/WARDuino #162: Connection time out in Latch

tolauwae, 2023-04-25:
Connecting in Latch works for v0.2.3 but times out on the develop branch.

Possible also an issue for the plugin, since largely the same code is used.

This issue blocks #158.

carllocos, 2023-05-10:
I indeed experienced the same issue on the plugin side but I experienced it on the plugin-version of the disco-plugin branch. However, I think that it still may be relevant.

In my case, the timeout occurred when the plugin refreshed its state for the first time just after flashing into the MCU i..e, when the plugin send an interrupt to the VM to request the state (in my case it was an inspect interrupt) that is displayed to the user via the GUI. Somehow the interrupt got received but the VM simply did not answer. I believe that the issue is that the VM was simply not ready to answer to the interrupt. Maybe when the VM got the interrupt the module was not fully loaded, making it impossible for the VM to answer.

On the disco-plugin branch, I had a temporary workaround via a timeout. I simply waited 2 extra seconds after flashing into the MCU. The bug did not manifest then. @tolauwae  Maybe you can also try that to verify whether that fixes your problem.

I recall talking about this with you but perhaps a new interrupt is needed to let the VM inform tool clients when the VM is ready? This is a better alternative to the timeout that I used.

tolauwae, 2023-05-11:
Yes this sounds like the same thing. So it is the first interrupt, just after starting the virtual machine locall? (emulator)

Because that is where my problem is too. Normally I wait until the string `"Listening"` is printed so I know that the TCP socket is live, but latch just keeps waiting for that message in the GitHub action.

tolauwae, 2023-06-05:
The issue was fixed with a simple flush of stdout after the `"Listening"` notice in the VM.

carllocos, 2023-06-12:
regarding https://github.com/TOPLLab/WARDuino/issues/162#issuecomment-1543598926

I experienced the issue when interacting with the MCU and not on the emulator. So what I have experienced is maybe still probably another issue.

commit d90485b9 (2023-06-05): Flush after "listening to socket" notice

Fixes #162
PR #184 (2023-06-14): Version 0.4.0
## What's Changed

### :boom: Breaking Changes

- Use virtual addresses in debug API

### :bug: Bug Fixes

- Add flush after "listening to socket" notice. Fixes #162 
- Make templatized Arduino compilation work on macOS

### :sparkles: Changes

- Add `inspect` command
- Extend `loadsnapshot` to load events and callbackmapping

### :hammer: Refactor

- Refactor dumps to use `inspect` command behind the scenes.

### :rotating_light: Testing

- Update compilation test to newer format
commit c43f5206 (2023-06-14): Flush after "listening to socket" notice

Fixes #162
closed 2023-06-14 by carllocos